xorbit/LiFePO4wered-Pi #1: [Feature Request] - Push button interaction instead of touch pad

biank88, 2016-10-04:
I've just got my module... it seems pretty good indeed.
Would it be possible to use a physical push button instead of the capacitive touch pad? 
Is there any particular reason why GPIO4 (pin_7) isn't wired to the uC?

xorbit, 2016-10-04:
The PCB currently does not have room to accommodate a button.  But it may be possible to add a surface mount footprint over the touch button.  It's a good idea, I'll keep it in mind for a future revision!  Biggest challenge may be to preserve as much ground plane as possible, it already is minimal.

That said, if you want to add a "remote" button that is mounted to a case or something (not on the PCB), that is already possible now.  For a remote touch pad, the easiest way is to solder a small wire to the "T" pad, and connect it to an area of copper foil or PCB copper.  You may need to play with the touch button settings using the lifepo4wered-cli tool to increase sensitivity.  If you want a physical button, you can wire a button in series with a small capacitor (20 - 100 pF should do) between the "T" and "G" pads.  You could possibly hack that on to the existing PCB as well, but you may need to scrape some of the solder mask off of the ground plane.

GPIO4 isn't wired to the micro because it isn't needed. :)  And routing was hard enough already.

biank88, 2016-10-04:
Thanks @xorbit ... I'll go for the physical push button in series with a capacitor. 
I'll need to move the led from the pcb to the case too. Thank you very much!

xorbit, 2016-10-04:
Good luck with your project! :)

closed 2016-10-04 by xorbit