xorbit/LiFePO4wered-Pi #3: Build fails on Raspbian PIXEL

montreuilg, 2016-10-20:
On a brand new Raspbian PIXEL install on RPI3, i get the following while trying to buuild from git:
`pi@dev:~/Documents/LiFePO4wered-Pi $ python build.py
gcc -c lifepo4wered-access.c -o build/DAEMON/lifepo4wered-access.o -std=c99 -Wall -O2
In file included from lifepo4wered-access.c:10:0:
/usr/include/linux/i2c-dev.h:37:8: error: redefinition of ‘struct i2c_msg’
 struct i2c_msg {
        ^
In file included from lifepo4wered-access.c:9:0:
/usr/include/linux/i2c.h:68:8: note: originally defined here
 struct i2c_msg {
        ^
In file included from lifepo4wered-access.c:10:0:
/usr/include/linux/i2c-dev.h:89:7: error: redefinition of ‘union i2c_smbus_data’
 union i2c_smbus_data {
       ^
In file included from lifepo4wered-access.c:9:0:
/usr/include/linux/i2c.h:128:7: note: originally defined here
 union i2c_smbus_data {
       ^
lifepo4wered-access.c: In function ‘read_lifepo4wered_data’:
lifepo4wered-access.c:59:16: warning: pointer targets in assignment differ in signedness [-Wpointer-sign]
   dread[0].buf = &reg;
                ^
lifepo4wered-access.c:64:16: warning: pointer targets in assignment differ in signedness [-Wpointer-sign]
   dread[1].buf = data;
                ^
lifepo4wered-access.c: In function ‘write_lifepo4wered_data’:
lifepo4wered-access.c:98:14: warning: pointer targets in assignment differ in signedness [-Wpointer-sign]
   dwrite.buf = payload;
              ^
fabricate: 'gcc' exited with status 1`

montreuilg, 2016-10-30:
Resolved, re-installed everything from scratch.

closed 2016-10-30 by montreuilg

schizobovine, 2017-09-25:
For those that run into the same issue, it was caused for me by having libi2c-dev installed. So, you can temporarily uninstall libi2c-dev, compile the LiFePO4wered-Pi tools, and then reinstall libi2c-dev, like so (from within the cloned repo):

```
sudo apt-get remove -y libi2c-dev
./build.py
sudo ./INSTALL.sh
sudo apt-get install -y libi2c-dev
```

This appears to be an upstream problem that's should be already fixed but may not have made it to Raspbian yet.

xorbit, 2017-09-25:
Thanks for reporting a solution for this, since I had not been able to duplicate it myself!