xorbit/LiFePO4wered-Pi #5: REQ: Autoshutdown

madeinoz67, 2016-12-14:
I'd like the ability to perform auto shutdown of Pi if incoming power is disconnected.  e.g. shutdown after x delay when incoming power in not present.

xorbit, 2016-12-14:
This is a feature that has been requested frequently, and as you can see in the code, it has been added to the host code already!  Look for AUTO_SHDN_TIME, which has been added to register version 4.  As this requires hardware support, it will first be available in the new LiFePO4wered/Pi3 that will hopefully be released later this week.  I'm hoping to backport this to future versions of the LiFePO4wered/Pi as well.

madeinoz67, 2017-02-21:
awesome, can the current version hardware be modded to take advantage of the autoshutdown?

xorbit, 2017-02-23:
Unfortunately not easily.  The original LiFePO4wered/Pi power manager is missing a connection from the 4-pin header to the microcontroller and the charger part doesn't have the correct resistive divider.

closed 2017-03-30 by xorbit