xorbit/LiFePO4wered-Pi #6: CLI doesn't seem to work

fiveseven808, 2017-04-08:
Here's what I get
```
$ lifepo4wered-cli get vbat
-1
$ lifepo4wered-cli get vout
-1
$ lifepo4wered-cli get VBAT
-1
$ lifepo4wered-cli get vin
-1
```

But I know the install must've been successful since I can now power on and shutdown my pi with the touch button just fine, The LED changes state when in bootup and shutdown process accordingly. 

Any help would be greatly appreciated. Thanks!

xorbit, 2017-04-08:
This may be a permissions issue.  Which distro are you using?  Does your user have permissions to access the `/dev/i2c-1` device?  Can you test if it works when you run the command as root or with `sudo`?

fiveseven808, 2017-04-08:
I truth with sudo, same response

fiveseven808, 2017-04-09:
I just tried it with a different SD card... same issue :\

fiveseven808, 2017-04-10:
What is the rage of i2c baudrates this thing an communicate at?

fiveseven808, 2017-04-10:
Seems that the CLI is a little flakey. Any ideas? 

```
$ lifepo4wered-cli set LED_STATE 3
3
$ lifepo4wered-cli set LED_STATE 1
-1
$ lifepo4wered-cli set LED_STATE 1
1
$ lifepo4wered-cli get vin
170
$ lifepo4wered-cli get vin
-1
$ lifepo4wered-cli get vin
-1
$ lifepo4wered-cli get vin
170
$ lifepo4wered-cli get vbat
-1
$ lifepo4wered-cli get vbat
3299
$ lifepo4wered-cli get vout
-1
$ lifepo4wered-cli get vout
-1
$ lifepo4wered-cli get vout
4914
$ lifepo4wered-cli get vbat_min
2854
```

xorbit, 2017-04-10:
The LiFePO4wered/Pi and /Pi3 are designed to run at the default I2C 100 kHz clock.  I have not tested it at higher clock speeds.  The MSP430G2231 has a limited I2C peripheral that requires a lot of software interaction, which limits the maximum speed.
Was the last log run at 100 kHz or less clock speed?  How much communication is there with other I2C peripherals like the screen you mention?  As you can see in `lifepo4wered-data.c`, I do 5 retries spaced 2 ms apart before I return with an error code.  If your I2C bus is very busy, you may need to alter the number of retries or the delay between the retries in this file, recompile and reinstall.  This is application specific, the default values work fine for most users as far as I know.

closed 2017-05-10 by xorbit