xorbit/LiFePO4wered-Pi #7: Feature Request: Power-on Timeout

fiveseven808, 2017-04-08:
Currently, if the device is turned on, without a pi connected, you have to pull the battery to turn the thing "off" 

I'd like to suggest a timeout setting where you can specify that if the device does not receive a "powered on" signal within 30 seconds to a minute or so (make adjustable?) then do a hard power-off of the device.

fiveseven808, 2017-04-08:
In addition to this, I'm not sure it's possible, but I'd like to request the ability to detect whether the pi is of or not (without receiving commands from the program)  and turn off. I.e. the pi shut itself down instead of set AUTO_SHDN_TIME?

xorbit, 2017-04-10:
I understand your first request and agree this would be a good feature to add.  I'd actually like to add a similar timeout to the shutdown, in case the Pi is not responding to a request to shut down.

I do not understand your second request.  The LiFePO4wered/Pi already knows the Pi has shut down by monitoring the TX line.  It also knows when the Pi is being shut down by any means, because the daemon informs it that this is happening.  Can you clarify?

xorbit, 2017-05-10:
Closed due to lack of clarification

closed 2017-05-10 by xorbit