xorbit/LiFePO4wered-Pi #12: Pi3 quesstion

mjmeans, 2017-11-15:
My application will use other Pi HAT's and a touch screen. Both receive their power from the GPIO pins. Will the LiFePO4wered/Pi3 actually cut off power to the GPIO pins so that the HAT and touch screen don't consume battery power while the Pi is shut down?

xorbit, 2017-11-16:
Yes, the LiFePO4wered/Pi3 applies power through the GPIO pins and after shutdown removes it, so anything powered through the GPIO pins should turn off at the same time.

mjmeans, 2017-11-17:
It appears that you use UART TXD to detect power off state. What is the behaviour when LiFePO4wered/Pi3 is being used and the user boots into NOOBS to reinstall the OS? During this time, there is no daemon running. How will LiFePO4wered/Pi3 react in this case? How will it react in the case that the user boots into NOOBS and NOOBS is running on an poor connection that would take hours to download an OS, or if NOBS is left running indefinitely?

My application will be using Windows 10 IoT Core and official Raspberry Pi 7" touch screen. Windows IoT Core takes much longer to boot than Raspbian, like 60 seconds as opposed to 5 seconds. In that way it is similar to NOOBS booting, but much faster than NOOBS installing a new OS. I can write a library for the interface if necessary, I've written Windows code for other products: UPS PIco HV3.0 Stack, ATXRaspi and Pi-UpTime UPS. None of those do what I want and I'm considering trying your design out if you think it might work. Porting over your daemon to Windows 10 IoT Core is not a problem for me.

xorbit, 2017-11-19:
Currently the behavior is that if there's no daemon, the LiFePO4wered/Pi will stay in the "booting" phase indefinitely.  Of course it won't be doing its job ensuring clean shutdown in that case.  I'm not necessarily considering the current behavior ideal, so in a future version of the firmware that I'm working on, I'm adding a settable timeout so that if the Pi daemon doesn't respond, eventually the Pi power will be removed.  You will still be able to turn the timeout off.  I will be recommending that people do system install without the LiFePO4wered/Pi (you're interacting with it anyway and can do a clean shutdown), but with the current firmware it will work like you want.

You should be able to write a driver for Windows IoT Core.  All the registers are documented in the manual.  All you need is I2C access and it should work.  You can use my Linux code as a guide.  Only access to the PI_RUNNING register is absolutely necessary.

xorbit, 2018-01-07:
The timeout for the boot and shutdown states has now been implemented, devices shipped since 2018 will have this feature active by default.

closed 2018-01-07 by xorbit