xorbit/LiFePO4wered-Pi #14: Can I update the firmware?

unphased, 2018-01-08:
I have in my possession a LiFePO4wered/Pi3 unit I bought in 2017 and does not have the fancy new 2018 release firmware. 

I'm hoping there is a not-so-difficult way to update the firmware on it!

Thank you.

xorbit, 2018-01-17:
Hello, sorry about my late response, I somehow missed this.

Unfortunately the LiFePO4wered/Pi3 is not made to be updated in the field.  One of the limitations of using a tiny micro: there's no room for update functionality.  Hopefully in a future version I may be able to add this.

I like your idea of trying to log in to the Pi with the UART to do a shutdown, it's an interesting concept.  But also not on the table with the current crop of devices unfortunately, but I'll have to keep it on my list of ideas for future devices!

closed 2018-04-23 by xorbit

aerickson, 2018-12-12:
If someone were very motivated, what would it take? Does the chip need to be desoldered?

I also have an older firmware and would love the 10-second button press to shutdown feature.

Can the new + model be upgraded via software?

Thanks!

unphased, 2018-12-12:
The procedure that I use in these situations (which is quite common in my workflow) is i use my pinky or a tool like a pen to push on the cell to disconnect it from the contacts. That effectively shuts down the lifepo4wered/pi3 unit. Due to it having a spring, this procedure does not result in damage.

xorbit, 2018-12-12:
The chip doesn't need to be desoldered, just reprogrammed.  If you want you can send me the unit and I can reprogram it for you if you're willing to pay for the return shipping.

The + model doesn't have field reprogrammability either, it's based on a micro in the same family.  It does have a little more flash, but not enough to do safe updates in the field.

aerickson, 2018-12-14:
@unphased Exactly what I ended up doing also (like 20 times the other day while troubleshooting kernel panics).

@xorbit Sweet. Just paypal $4 to your email and include a note with the package? 

Do you have any experience with the relatively cheap MSP 430 programmers (I found MSP LSD-FET430UIF on ebay/aliexpress)? Would it work? Not opposed to adding some tools to my collection and learning. It's programmed via 2-wire (pins 16 and 17)?

Maybe in the ++ model. :)

xorbit, 2018-12-18:
Yes, that's right, $4 if you're in the US that is. :)  Otherwise it's $14 for shipping.

I haven't tried the cheap MSP430 programmers, either way I don't distribute the firmware binaries at the moment.  Hopefully in the future there may be an updatable version, but it's a tricky problem to get right when you need to update and reboot the device that's powering the system!