xorbit/LiFePO4wered-Pi #16: Question: Why avoid_warnings=2?

unphased, 2018-01-12:
Enabling turbo in low voltage condition may lead to data corruption, which runs counter to the purpose of a UPS. 

Curious about the reasoning.

xorbit, 2018-01-13:
Good catch.  That ended up there for historical reasons.  My first prototype of the original LiFePO4wered/Pi provided the battery voltage directly to the Pi, without a 5V boost converter.  So the under voltage warning was always there, which was annoying.  I have only produced versions with 5V boost converter commercially, so this isn't necessary and probably shouldn't be there.
I'll remove it in a future update.

unphased, 2018-01-17:
Amazing, were you using a LiFe cell as well? I guess it isn't so surprising that the Pi can survive on much lower than 5 volts. Now you're making me wonder if it would make sense to directly hook the LiFe cell to 3.3v input on the Pi as the operating voltage happens to land right around there.

xorbit, 2018-01-17:
Yes, it worked fine powered directly from a LiFePO4 cell.  For its internal use the Pi just converts the 5V to 3.3V/2.5V/1.8V anyway.  Other than that it just puts the 5V on the USB ports (and most USB devices convert to lower voltage internally as well).  If I remember correctly I could run a Pi with WiFi dongle reliably down to around 2.9V, at which point the LiFePO4 is pretty much empty anyway.

I think hooking a LiFePO4 up to the Pi's 3.3V might burn out a regulator on the Pi.  I remember trying that, and I remember burning out the regulator, but that was on the original Model B which used a linear regulator.  I guess if you can spare a Pi it might be worth the try to power the 3.3V and see what happens. :)  Otherwise, I was powering the 5V rail from the LiFePO4 cell and since the newer Pi's use a switch mode regulator, if the input is low enough it will just turn on the pass transistor which hardly causes any drop.

xorbit, 2018-10-19:
Removed avoid_warnings=2 from the install script.

closed 2018-10-19 by xorbit