xorbit/LiFePO4wered-Pi #17: Clarity of doc around use of UART

unphased, 2018-01-12:
I see now that the installer ensures `enable_uart=1` so that it turns on so it can be used to communicate the shutdown state. 

In various comments you mention only the TX is used.

It was unclear for me until I did some digging: The LiFePO4wered-Pi does NOT do any actual "reading" on the TX. It only checks if it is high. Yes you have stated this, but I was worried about it upon first seeing this.

Now that I know that the installer only sets `enable_uart=1`, I know my software will be completely unaffected by the presence of the LiFePO4wered-Pi also being attached to the pin. Maybe you can add a sentence in the product brief that clarifies this.

xorbit, 2018-01-13:
Good suggestion, thanks!

xorbit, 2018-04-23:
The latest product briefs have been updated with detailed information on how each pin is used.  See https://lifepo4wered.com/files/LiFePO4wered-Pi3-Product-Brief.pdf

closed 2018-04-23 by xorbit