xorbit/LiFePO4wered-Pi #19: Best configuration?

rwood, 2018-04-10:
I have an arcade box that I built that runs off a rpi3.  The kids keep just shutting off the power and can't seem to grasp the concept of shutting down the system first.  I discovered the LiFePo4wered and the clouds parted.  I've installed it and set the auto_boot to 1.  But when the power gets cut the rpi will still run for a while before shutting down.  What's the best way to configure this to automatically shutdown a few seconds after power has been cut?

xorbit, 2018-04-10:
Do you have a LiFePO4wered/Pi or LiFePO4wered/Pi3?  The LiFePO4wered/Pi3 can sense the presence or absence of input power, the LiFePO4wered/Pi lacks the necessary circuitry for this function.  If you have a LiFePO4wered/Pi3 you can set AUTO_BOOT to 3 or 4 (depending on whether you still want to be able to turn the unit off with the touch button) and AUTO_SHDN_TIME to 0 for immediate shutdown when external power is removed or a number of minutes if you want it to run through short power outages.  These settings will cause automatic boot and shutdown based on presence of external power.

rwood, 2018-04-10:
It's a LiFePO4wered/Pi3.  AUTO_BOOT 3 and AUTO_SHDN_TIME 1 work great.  Thanks.

closed 2018-04-10 by rwood