xorbit/LiFePO4wered-Pi #20: Compilation problems

luiss, 2018-05-16:
Hi, I'm on a Rpi with Debian 3.18.5-1~exp1+rpi19 (2015-08-08) armv7l GNU/Linux.
When I first tried to compile the code I have got a bunch of errors like:
 `storage size of `ts' isn't known`
and 
`undefined reference to `le32toh'`.

The first I solved by adding the string 
 `#define _POSIX_C_SOURCE 199309L`at the beginning of to the lifepo4wered-daemon.c 
see: http://gcc-help.gcc.gnu.narkive.com/8xCaKI6r/problem-with-struct-timespec-and-c99-standard

The second was solved by exchanging the line:
`#define _DEFAULT_SOURCE` to `#define _BSD_SOURCE`.
see https://stackoverflow.com/questions/26713198/undefined-reference-to-htole32

You may add this somewhere in the README for example to help others on the same issues.
Best regards.

xorbit, 2018-06-04:
Hi Luis,

Is that on Debian Wheezy?  Or Jessie?
I know I used to have `#define _BSD_SOURCE` but I needed to change that to `#define _DEFAULT_SOURCE` to work on more recent releases.
Is there a reason to be on an old distro?

luiss, 2018-06-05:
Hi Patrick,
May be the problem comes from an error on my part for trying to use the LifePo4wered on an Pi 2 Model B, where in fact I should have been using the LifePo4wered/Pi3?
However, if I understand well the LifePo4wered is supposed to work (as a battery but not as an UPS) for the new Rpi models thus it seems to me that from this point of view it is a concern.

I have no access today to my module so I will provide the details of the distribution version as soon as possible.

xorbit, 2018-06-06:
No, software wise it makes no difference which LiFePO4wered device you use, they all use the same host software.
As I said, things change from one release to the next with what #define should be used, etc.  I try to support the last two major releases (stretch and jessie at the moment).  Anything older is not officially supported, but I try not to break things if not necessary of course.

luiss, 2018-06-06:
Hi,
You are right, I'm on Wheezy. 
Despite a few dist-upgrade in the past I may be a couple of releases old.
May the issue be considered as solved?
Luiss

closed 2018-06-08 by luiss