xorbit/LiFePO4wered-Pi #24: Pi doesn't seem to shutdown seems to be in loop

bbourqu, 2018-11-13:
```
[ ~ ]$ lifepo4wered-cli get AUTO_BOOT     
4
[~ ]$ lifepo4wered-cli get AUTO_SHDN_TIME
0
```
When I unplug from the powersource the `PWR LED` pulses and I can see the activity LED on the Pi show like its in its shutdown phase but the power LED never shuts off and the activity LED continues to blink. The `PWR LED` never stops pulsing.  No SSH access until I plug the power supply back in.  The fan I have connected to the VOUT pads also continues to run.  Holding down the switch for 10 seconds does shut off the pi.  

Is there anything I'm missing that might create an issue?  The scenario I'd like to have is that when main power is removed, then the pi is gracefully shutdown, on main power return the pi is booted back up.

Raspberry Pi 3 B+ running  Raspbian Stretch.

xorbit, 2018-11-14:
Hi,
That is the correct config to get the behavior you want.  Which LiFePO4wered model do you have and when did you get it?

bbourqu, 2018-11-15:
LiFePO4wered/Pi+ (14500 Battery) non-stackable headers.  About 13 days ago(Shipped Nov, 1st)

xorbit, 2018-11-15:
- Is there anything else connected to the GPIOs?

- Does your boot.txt contain these lines:

# I2C device tree settings
dtparam=i2c1=on
dtparam=i2c_arm=on

# UART device tree settings
enable_uart=1

- Do you have a way to check whether UART TX (GPIO pin 8) goes low when the Pi is done shutting down?

If for some reason the TX line is not going low or it is not being detected, there are a couple of workarounds:

- You can switch from shutdown detection by TX monitoring to IOUT monitoring by using the IOUT_SHDN_THRESH register.  Documented on page 24 of the product brief.

- You can not bother with shutdown detection and switch power off based on time by adjusting the PI_SHDN_TO register.  Documented on page 28 of the product brief.

https://lifepo4wered.com/files/LiFePO4wered-Pi+-Product-Brief.pdf

bbourqu, 2018-11-16:
* Nothing connected to the GPIOs besides the LiFePo4wered module
* `/boot/config.txt` contains the following:
```# I2C device tree settings
dtparam=i2c1=on
dtparam=i2c_arm=on

# UART device tree settings
enable_uart=1
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
overscan_scale=1
```

* I don't have a way currently to check that pin

I'll try the first option and report back

bbourqu, 2018-11-16:
Before trying the first option I pulled the power again.  This time the pi shutdown.  Plugged the power back in, power came back on.

Tried again, pi shutdown, plugged in pi came back on.  Tried one more time and this time I got back into the power cycle loop.  

Let the Pi run for about ten minutes and tried the cycle again, same thing about every third time the pi gets into the loop.  I plug in and unplug over maybe a two minute period.

bbourqu, 2018-11-20:
Lets go ahead and close this issue.  I got the capacitor but totally messed up the pads de-soldering and soldering back.  I'll look at purchasing the correct version at a future date.

closed 2018-11-20 by bbourqu

openedhardware, 2020-05-11:
I do have the same issue.. RPi 3B+ and 14500 battery.

@xorbit 

What is a desired value for `IOUT_SHDN_THRESH` register?

Otherwise, where should I add the bulk capacitor?

Thanks!

xorbit, 2021-01-11:
@openedhardware You need to determine `IOUT_SHDN_THRESH` experimentally for your system.  Get the value of IOUT and go from there.  That said, I think a value of about 120 (mA) may work if you don't have external (USB) loads etc.  Note that you should only use this if the TX line shutdown detection doesn't work for you for some reason.

Bulk capacitor can be added to the VOUT pads.