xorbit/LiFePO4wered-Pi #27: Nice, but it would be nice to have loss of external power logged

garethhowell, 2019-05-17:
The main daemon only does something when the battery expires and the system needs to be shutdown. It would be nice if there was some way of signalling that external power was lost and we are now running on battery: i.e. something more akin to the level of information you get from something like a APC UPS

xorbit, 2019-05-17:
The daemon provides the minimum functionality to make the device work as expected for most people.  If you want more functionality than that you can add what you want to the daemon yourself, or use the provided tools to implement the behavior in your own separate programs or scripts.

For instance you can query `lifepo4wered-cli get vin` to see if external power is lost and take appropriate action from a shell script.  Or use the C, Python or Node.js bindings to do it.

garethhowell, 2019-05-18:
Fair point, Patrick.
I’ll see what I can do. It’s been a (very) long time since I did any C
programming (about 50 years I think), but I’ll give some thought to forking
the code and having a go.

Gareth

ristomatti, 2019-05-18:
I gave this issue thumbs up but also understand the design philosophy. @garethhowell I think it'll be far easier to write a simple shell script to do this. Just request the value once a minute for example and either log the event where you want. If you create a systemd service definition file to keep it running, you can also pass the output to a system log if that is what you're after.

garethhowell, 2019-05-19:
That makes sense, @ristomatti. I’ve done a few system daemons in python, so
I’ll use that.

Gareth

closed 2020-08-13 by xorbit