xorbit/LiFePO4wered-Pi #45: LiFePO4wered does not react

adoconnection, 2020-09-03:
Hi, im using several of your boards and i like it!
However there are two beards that does strange things: not matter what i do it never does anything.

https://www.youtube.com/watch?v=ia9m3pTtxas
first board works as expected, the second one not. Do you have any Ideas?

If I plug power directly into raspberry (3b+) I can communicated with board
pi@Golf:~ $ lifepo4wered-cli get
I2C_REG_VER = 7
I2C_ADDRESS = 67
LED_STATE = 1
TOUCH_STATE = 0
TOUCH_CAP_CYCLES = 0
TOUCH_THRESHOLD = 12
TOUCH_HYSTERESIS = 2
DCO_RSEL = 14
DCO_DCOMOD = 171
VIN = 0
VBAT = 0
VOUT = 0
IOUT = 0
VBAT_MIN = 2850
VBAT_SHDN = 2950
VBAT_BOOT = 3150
VOUT_MAX = 3500
VIN_THRESHOLD = 4498
IOUT_SHDN_THRESHOLD = 0
VBAT_OFFSET = 49
VOUT_OFFSET = 26
VIN_OFFSET = 25
IOUT_OFFSET = 0
AUTO_BOOT = 3
WAKE_TIME = 0
SHDN_DELAY = 40
AUTO_SHDN_TIME = 0
PI_BOOT_TO = 300
PI_SHDN_TO = 10
RTC_TIME = 0
RTC_WAKE_TIME = 0
WATCHDOG_CFG = 0
WATCHDOG_GRACE = 20
WATCHDOG_TIMER = 0
PI_RUNNING = 1
CFG_WRITE = 70

xorbit, 2020-09-04:
Hi Alexander,

Ok, this is a really odd one.  Thanks for the data dump, it's very helpful.

It looks to me like the main loop in the firmware is stuck, trying to write the config to flash.  It would explain the non-responsiveness of the button, and the lack of voltage measurement values in the dump.  And it's the only thing that explains the value of 70 for CFG_WRITE.  You write 70 to this register to write to flash, but after the flash write is done, the code clears the register back to zero.

The situation isn't completely clear to me because it looks like I disable interrupts during flash write, but interrupts must be working to read data from I2C.

What I would suggest you try is remove external power and pop the battery out for 10 minutes (yes that long, need to make sure all internal nodes are discharged and it can take a while because of ultra low power).  Then put it back and see if it works.  If you had configured your unit and written config to flash, the evidence suggests this write may have failed, so configure and write to flash again to be safe.

adoconnection, 2020-09-05:
Thanks, will try

adoconnection, 2020-09-07:
Removing battery for a longer period did the trick for both boards: rev5 and older one. 
cool!

closed 2020-09-07 by adoconnection