xorbit/LiFePO4wered-Pi #49: Can LiFePO4wered-Pi board handle huge battery packs up to 100ah?

adoconnection, 2021-01-11:
There is a BAT (+-) pads ontop of the board, but are they capable of charging really big battery packs, lick 100Ah?

[image]

ristomatti, 2021-01-11:
I'm not an EE but I couldn't see why not. It would just take a VERY long time to charge if its empty. I would suggest precharging the battery with a beefier charger if you have one available.

xorbit, 2021-01-11:
@adoconnection Yes, like @ristomatti says, it would just take quite a while to charge the whole battery.

Whether is makes sense or not depends more on the load than the battery size.  As long as you overall get more energy in than out, it will work and the large battery gives you a nice long backup time.  But if you'd get to a situation where you actually discharge a lot of the battery capacity, it would be good if your incoming energy was significantly higher than your outgoing, to recover quickly.  I don't know if you're doing a battery powered application with off-line charging or something like a solar based system.

At 1.5A charge current, it would take about three days to completely recharge this size battery with the LiFePOwered/Pi+.

adoconnection, 2021-01-11:
Its a cycle charge/discharge: I charge lead acid battery with 220v ACDC, then this battery power smaller 5v devices in a field.
Thought would it work if I were to replace LeadAcid battery with bigger LiFePO4

> At 1.5A charge current, it would take about three days to completely recharge this size battery with the LiFePOwered/Pi+.

I see, thanks for fast reply 👍

xorbit, 2021-01-11:
If you use an external high power LiFePO4 charger, it should work fine to recharge the battery quickly.  You can even do it while the LiFePO4wered/Pi+ is connected, no need to take things apart.  The only thing you need to avoid is that your external charger should not be charging at the same time as the LiFePO4wered/Pi+ charger.  Just leave the LiFePO4wered/Pi+ input unconnected while the external charger is doing its thing.

adoconnection, 2021-01-12:
It wont work for me since everyting is always connected to each other. Well if you at some point decide to update the board, consider more Amps for charger as a feature request :)

xorbit, 2021-01-13:
@adoconnection I don't see why you conclude it wouldn't work.  Like I said: as long as you don't connect a input to the LiFePO4wered/Pi+, but just use an external charger connected to the LiFePO4 cell, it should work fine.  Just ignore the fact that the LiFePO4wered/Pi+ has a charger and you can still use all the power manager features, safe shutdown, etc.

adoconnection, 2021-01-14:
At first I thought you asking to plug in and out LiFePo4wered input every time before and after charging. Now I see what you mean. LiFePo4wered IN is never used and the board is always connected to a battery only. Thank you Patrick!

closed 2021-01-18 by adoconnection