xorbit/LiFePO4wered-Pi #50: 14500 model not using all the available power

adoconnection, 2021-01-25:
Hi, need your help again :)

Im using Pi3 B+ and 14500 Battery model.
Today I powered  LiFePO4wered from my new power source and it turned out LiFePO4wered is limiting input current to 1 amp, so the Pi will not boot until battery is charged well.
Isn't it supposed to take at least 2 amps? (my pi consumes 1.4 amps)

[image]

You may probably ask why im using 14500 and not 18650 as manual suggests. Because after using LiFePO4wered for more than 1 year only now I found out there is a small line in manual that states 14500 does not support 3B+ models.
I buy shields from crowd supply and had no idea about that. Please fix the description!

[image]
Its not only capacity difference but also Pi model itself
For USP I decided small battery is enough.

ristomatti, 2021-01-25:
I'm only another customer following the repo but you sent a photo of a power supply? It can only show the charge current which is typically smaller than the discharge current. You would need an electronic load (dummy load) to test the maximum discharge rate.

Also Raspberry Pi 3B+ should be taking just around that much current in case your photo shows the power supply powering the hat with a fully charged battery and Pi attached.

[image]

Source: https://raspi.tv/2018/how-much-power-does-raspberry-pi-3b-use-power-measurements

ristomatti, 2021-01-25:
Further check this datasheet which happened to be the first search hit for "14500 lifepo4 discharge rate": https://lifepo4wered.com/files/IFR14500EC-600mAh.pdf

You'll see the nominal capacity is 600mA and the maximum continous discharge rate is 2C which translates to 1.2A. It's a simple multiplication but you can verify it here http://www.batteriesinaflash.com/c-rating-calculator.

Of course this doesn't answer your comment on misleading marketing :zipper_mouth_face:. When I bought my LiFePO4wered hats it was said the 14500 version is only supposed to be used on a Pi Zero.

adoconnection, 2021-01-25:
> It can only show the charge current which is typically smaller than the discharge current

is there a typo? charge current should be bigger than discharge current, otherwise the LiFePO4wered will not be able to power pi and charge battery at the same time.

**UPD**
I changed LiFePO4wered  input voltage to 12v and consumed current is still 0.4-0.5A but the system now boots fine with all the hats and LTE modem pugged into PI, looks like a solution to me.

xorbit, 2021-01-25:
Sorry if this comes over as rude, but it's not my problem if people don't read. :-/
It's actually a pet peeve of mine, don't take it personal.  But I can write all I want and it's of no use if people don't read it.

You include a screen shot of the Crowd Supply site, claiming misleading marketing.  But if you'd just scroll down you'd find this section:

> Compatibility
>
>The LiFePO4wered/Pi+ with 18650 battery is compatible with all modern, genuine Raspberry Pi products (Model A+, Model B+, 2, 3, 3 Model B+, Zero, and Zero W). The version with 14500 size battery is compatible with all of those, except the Pi 3 Model B+. LiFePO4wered/Pi+ can be used with the original Raspberry Pi Model A and B boards if you use extra wire or remove the RCA connector from the boards. In addition, we’ve confirmed that LiFePO4wered/Pi+ is compatible with the following non-Raspberry Pi single-board computers: Odroid C2, Nano Pi S2, Allwinner H3, and ASUS Tinker Board. The LiFePO4wered/Pi+ was designed to work with most single-board computers with a Raspberry Pi-compatible GPIO footprint. High power boards may only work with the 18650 battery, and modification to heat sinks or using extra headers for spacing may be necessary.

There's also a 37 page Product Brief here: https://lifepo4wered.com/files/LiFePO4wered-Pi+-Product-Brief.pdf
On page 12 starts a nearly 3 page "limitations" section, talking about some of these issues, including for the Pi 4 which didn't exist yet during the Crowd Supply campaign.

But you complain that the individual product option selection on Crowd Supply (not even under my control) doesn't tell you all the details.  How much of those 37 pages should be crammed in them?

The limitations section also talks about how the external supply and wiring can influence how much current is drawn if there's voltage drop.  Sorry but I'm not going to repeat it all here.  But it's likely why it works better for you at 12V.

xorbit, 2021-01-25:
Because this happens so often, I'm mostly phasing out the 14500 battery version.  The amount of support work generated because people don't bother to read is just too much.  It will only be available as special order in volume in the future.

adoconnection, 2021-01-25:
First of all I love your product. Im sorry if I hurt your feelings, dont get me wrong, im looking for help only.

> How much of those 37 pages should be crammed in them?

"One does not have to be writer to post critics". I have no idea, its not my product, but im happy to help with ideas. Clearly page 12 is not appropriate place for information of such importance.

So it happened Im good at UX so I'd place a table right below welcome image, no text is required at all and it is easy to see there are some limitations

[image]

this kind of infographics allow you to place more relevant info if needed

[image]

adoconnection, 2021-01-25:
If im allowed to get back to my question, Am I ok with 12v input if im using LiFePO4wered as UPS? 
And Im still confused why LiFePO4wered  consume only 0,5ams at 5v input voltage while manual suggests to use 3amps external power.

adoconnection, 2021-01-25:
I think this lines requires your attention

[image]

[image]

xorbit, 2021-01-25:
I like the way your graphics look, but there are a couple of problems:

- The next person who comes along would prefer to have some other info they found more important on the first page.
- People wouldn't necessarily know what the two different pictures mean, that has not been explained yet.
- Like most infographics, it attempts to condense a complicated situation into a simple graphic and as a result conveys inaccurate information.

It kind of shows by the various things you have underlined, thinking that you spotted inconsistencies in the docs.  Unfortunately, I've checked again and every one of those statements is correct!  I can't help it that it's a complicated engineering situation that depends a lot on various factors.  My "simplified version" is: in the section specifically about the 14500 version I say: "It does not support the Raspberry Pi Model 3 A+ and B+ or the Raspberry Pi 4."

The only way I see to simplify this is by getting rid of the 14500 version and the complications it causes in the docs.

To come back to your original question: the system only draws (or tries to draw) what it needs.  If you have no load, it only needs to charge the battery and the battery only accepts ~0.5A max.  When you have an additional load (like a running Pi) it should draw more.  If it doesn't, most likely the draw is causing voltage sag across the cable and this causes the input current to be limited.  This is described under limitations.  If you increase the input voltage (and don't add a MPP resistor), you will not run into this limit so easily, that's why in your situation it works with 12V but not 5V.

Bottom line: I have tested that the 14500 version works with 2A load when plugged in to a good 3A supply with a good USB cable.  If it doesn't for you, your supply or cable is most likely to blame.  But once unplugged, the higher power path resistance for the 14500 battery prevents it from backing up more current then 0.75A.  It may last enough to trigger a clean shutdown but I can't guarantee that so I say "don't use it with a Pi3+ or Pi4.

But here I am again, having spent several hours on supporting the 14500 version because it's confusing and just not worth the amount of support time I have to spend on it.

closed 2021-05-22 by xorbit