xorbit/LiFePO4wered-Pi #52: Solar panel question

scottsweb, 2021-07-30:
I am working on a little project with my LiFePO4wered-Pi board and noticed in the documentation a resistor is recommended to customise the MPP setting for the panel:

```
RMPP = 51815 / (VMPP – 4.66)
```

I contacted the panel manufacturer and they came back with a value of `Vmmp: 7.38V`. I just wanted to check that the recommendation is a: 19049 ohm / 19 kohm resistor on these pads. 

I only have a 20k resistor to hand, would that still be suitable or should I find something a little closer?

mstaack, 2021-07-30:
Should be fine

xorbit, 2021-07-30:
That will give you an MPP voltage of 7.25V, which should be just fine.  Err on the lower-than-spec Vmpp side, it will work better in low light and the power drop off is slower on the low side.
Note that when you add this resistor the unit won't charge from voltages lower than Vmpp anymore, so 5V USB charging won't work after you add it.

scottsweb, 2021-07-31:
Thank you both for the clarification, that is a huge help. I would like to keep the 5V USB charging for now so I will save adding the resistor until I am sure we get enough Swedish sun 😄

closed 2021-08-02 by xorbit