xorbit/LiFePO4wered-Pi #53: Is there a 3D STEP file available for this device?

madeinoz67, 2021-08-16:
Hi there,

I need to design a case for my current project that incorporates one of these, is there a 3D STEP or Freecad file available?

Thanks

xorbit, 2021-08-16:
Hi Stephen,

Unfortunately not.  Several people with the necessary skills have said they would make a model and share it with me to share with customers, but up till now it hasn't happened.

Patrick

rcrouthamel, 2021-08-16:
[LiFePO4wered-PiPlus18650-Unofficial.zip](https://github.com/xorbit/LiFePO4wered-Pi/files/6994467/LiFePO4wered-PiPlus18650-Unofficial.zip)
Here's a .STEP file of the 18650 model with stacking header I created for mechanical integration.  This is a simplified model and may not use the exact same components used by Patrick, but should be sufficient for most uses.  It is provided as-is with no warranty.  Patrick, feel free to use as you see fit.

Rob

xorbit, 2021-08-16:
Thank you very much Rob, much appreciated!  Does it solve your problem Stephen?

madeinoz67, 2021-08-17:
awesome @rcrouthamel I only need a simplified model and is perfect for what I want

closed 2021-08-17 by madeinoz67

madeinoz67, 2021-08-17:
@xorbit or @rcrouthamel can I suggest putting it up on GrabCad so its easy to find?