xorbit/LiFePO4wered-Pi #55: Taking less power from charging USB port than the base Pi does.

SharkWipf, 2022-01-12:
I just got my LiFePO4wered/Pi+ today, mostly to absorb peaks in power draw.
However, currently, it seems to discharge faster than it charges on a power supply that _should_ be able to handle it fine.
While my Pi3B by itself can run on this port unthrottled and without voltage warnings, while sustaining a high CPU and I/O load and having HDMI and USB devices running, if I run through the LiFePO4wered/Pi+, it discharges faster than it can charge even while manually throttling (600MHz), idle, no HDMI and no USB devices. Same cable and everything.

Looking at the lifepo4wered-cli output, I see vbat slowly dropping, while vin sits at 4858, and vout at around 5000.

Could it be the LiFePO4wered/Pi+ is not taking advantage of the full capacity of this USB port, whereas the "bare" Pi3B is?
The port in question is the special USB charging port on the nuc6i5syh, I'm unsure of the exact specifications of this port, all I know is that it was capable of driving this Pi3B unthrottled with a lot more load fine.

I should be getting the official Pi charger tomorrow, I assume everything will be fine with that charger, but I would expect it to work fine with this charge port as well, since it's, as far as I can tell, more than powerful enough for the bare Pi3B.

xorbit, 2022-01-12:
Hi, which version do you have, the one with blue 14500 (AA) size battery or the 18650 battery?

SharkWipf, 2022-01-12:
It's the 18650 version, [from Mouser](https://www.mouser.com/ProductDetail/silicognition/cs-lifepo-02/?qs=u16ybLDytRbGKL6uzZpvwA%3d%3d&countrycode=DE&currencycode=EUR).

xorbit, 2022-01-12:
The input voltage reported by the unit is kind of low, it's possible it's limiting the charge current because of that.

The Pi on its own draws whatever it needs, whether the power source sags or not.  Are you getting any low voltage warnings if you power the Pi directly?

The LiFePO4wered/Pi+ on the other hand uses a MPP (maximum power point) input stage so it can be used with solar, but this is always active and prevents the unit from overloading weak power sources.  It will basically limit current draw if the input voltage sags.  This can be due to a weak source or a long or thin gauge cable that drops voltage.

Try it with the Pi charger and maybe a different cable too, see if you can get the Vin to report closer to 5V (5000) and see if there still is a problem then.

SharkWipf, 2022-01-12:
Ah, my bad, I just checked with `vcgencmd`, without the LiFePO4wered/Pi+, and it does tell me it's undervolted: 
```
$ vcgencmd get_throttled
throttled=0x50005
```

I guess I was looking for the wrong signs, and the charge port really isn't sufficient after all. I'll just have to wait for the official power supply tomorrow I guess. Thanks for your time either way!

closed 2022-01-12 by SharkWipf