xorbit/LiFePO4wered-Pi #57: Reboot of the RPI blocked with the LiFePO4wered/Pi+ abd CLI

Lunik971, 2022-03-05:
hello,

I have a issue with the LiFePO4wered/Pi+ and the programm, when I send a reboot command the system just shutdown and do not comme up.
Does I have a setting to chagne for that ?

Issue :
1 : Send Reboot command
2 : The system power off
3 : The RPI never power on, need to press the button to power on again.

xorbit, 2022-03-07:
Most likely the issue can be fixed using the SHDN_DELAY register.  The problem most likely is that the newer Pis take much more time in the bootloader than old Pis.  The default value for SHDN_DELAY was appropriate for the older Pis but is too short for the newer ones.  The delay basically sets a wait time after the kernel shut down before power is removed, and a reboot can set TX high again to keep the system running through a reboot, if it waits long enough.

If you run `lifepo4wered-cli get SHDN_DELAY` you can see the current value of the register.  Most likely it's the old default value of 40.  Use `lifepo4wered-cli set SHDN_DELAY 64` to set it to the new higher default value and see if that fixes the issue.  If you want to make it permanent by writing it to flash, also run `lifepo4wered-cli set CFG_WRITE 70`.

Lunik971, 2022-03-14:
Hello,
Sorry for the late answer, ans yes this solved m'y issue.
Many thanks

closed 2022-03-14 by Lunik971