xorbit/LiFePO4wered-Pi #59: Question: Voltage across VIN pins?

scottsweb, 2022-05-19:
A quick question for you @xorbit ... I am measuring the voltage out of my solar panel at around 6.5v but when I measure across the VIN pins, I only get around 4.6v. Should I expect to get the same voltage there? and if so I am guessing that indicates I may need some better wiring between my panel and the board. I haven't yet soldered the resistor we spoke about in #52, so I also wondered if that could be related?

From the software I noticed my VIN sits around 4.6v and VOUT at around 4.9v-5v, so my battery is never getting topped off enough to sustain the Pi longer term.

xorbit, 2022-05-19:
I assume the first measurement is without the solar panel connected (open circuit, Voc), while the second is when connected to the board?  In that case, what you see is totally expected because without MPP resistor added, the Vmpp is 4.66V nominal.  Meaning the circuit will not draw more current if it makes the input voltage fall below this voltage.  So your battery charging may not keep up with the load.

You should probably read up on how solar panels work to have a better chance at success.  The MPP resistor pads are there for a reason if you want to extract the most energy from the panel.  See page 17 of the [manual](https://lifepo4wered.com/files/LiFePO4wered-Pi+-Product-Brief.pdf).  But seeing that Voc is 6.5 V, 4.6 V is probably not too far off from the panel's Vmpp (likely 5-ish V), so it's probable the panel is under sized to maintain a Pi long term.  If you don't need to have the system up continuously, using the wake timer and having the Pi shut off some percentage of the time is a good way to reduce power use and make it work with a small panel.  If your system needs to be up 24/7 then (subject to the Pi you use, the system load and any connected peripherals) a 20W panel may be a good starting point.

What is the wattage of the current panel you use?

scottsweb, 2022-05-19:
Thanks for the reply.

> I assume the first measurement is without the solar panel connected (open circuit, Voc), while the second is when connected to the board?

The first measurement was with the panel connect to VIN and measuring across those pads, the second measurement was with the panel unplugged and just measuring the direct output from it. 

It sounds like the 4.66v is expected though from your description. 

This is the panel: https://voltaicsystems.com/9-watt-panel/ - its 9 watts. The manufacturer said the Vmmp is `7.38V`. So I should be able to add a resistor and get some slightly better results I think.

The system doesn't have to be up continuiously but I would like to make sure I have optmised for the maximum ammount of charge... ideally when the Pi is on, I would like there to be enough power to charge the battery and keep the Pi running. I am using the original Pi Zero and it draws about `120 mA` when under some load, normally sits around `100 mA`.

Perhaps I just need to find a bigger pannel 🤔

xorbit, 2022-05-20:
For a Vmpp of 7.38 V you definitely want to have the MPP resistor, allowing the voltage to drop to 4.6 V will very much limit the amount of power you can extract.  Note that once this resistor is installed, the unit will not charge from USB anymore because the USB voltage is below the MPP voltage.

9W is not enough to run 24/7.  It may be just enough to only run during daytime, but not day and night.  This was my experience with a 10W panel.

scottsweb, 2022-05-21:
Great stuff, thanks again for the info. I have just soldered my resistor in place and will see how it goes. It is cloudy today but I am still reading more voltage across VIN, at about `5.5v`. 

> 9W is not enough to run 24/7. It may be just enough to only run during daytime, but not day and night. This was my experience with a 10W panel.

My 'trick' here is to use a massive battery (`20Ah`) in order to give me about a week of run time without any sun. Longer term though it sounds like I will be better off with a higer wattage panel. [Perhaps something like this 20w one](https://voltaicsystems.com/20-watt-6-volt-solar-panel-etfe/).

Happy to close this issue now. Thanks again!

xorbit, 2022-05-21:
From my experience (with a Model A, not a Pi Zero), even with continuously sunny days 9W is most likely not enough to run the Pi 24/7 and eventually that large battery will run out.  My test was with a 6Ah battery.  YMMV 🤷‍♂️

scottsweb, 2022-05-21:
Noted, I will see if I can find a new panel 🌞

closed 2022-05-21 by scottsweb