xorbit/LiFePO4wered-Pi #60: Register corruption causing shutdown and no-boot

DAVe3283, 2022-07-05:
I am experiencing issues nearly identical to #48. There is nothing on the GPIO headers except the LiFePO4wered/Pi3. I am running Raspberry Pi OS 2022-04-04, fully patched.

I am running `lifepo4wered-cli get` every 60 seconds to log statistics to Home Assistant. But when I run `lifepo4wered-cli get` over and over, things get corrupted. Here is what happened last night:

- WAKE_TIME started at 0 then went to 255 (0x00FF) at 19:36, then to 65415 (0xFF87) at 21:04.
- TOUCH_STATE went from 0 to 15 (0x0F) at 03:13, likely causing the service to shut down the Pi. The button was not actually touched.
- PI_RUNNING went from 1 to 135 (0x87) at 14:36, then to 255 (0xFF) at 21:04, then back to 135 (0x87) at 21:07, and finally to 0 at 3:13.

All the measured voltages also started reporting garbage at 01:50 this morning:
- [image]
- [image]
- [image]

I don't believe any of those voltages actually changed, as the battery can't jump 1.3 volts in 1 minute, so it is likely the calibration/offset getting corrupted.

If just the daemon is run, the problem doesn't appear to show up (at least not quickly). But when gathering stats, within a day things corrupt and the Pi turns off. And due to WAKE_TIME getting set stupid high, I have to remove the battery to reset the MCU before I can turn the Pi back on.

xorbit, 2022-07-05:
Yeah, those voltage jumps obviously aren't real.

In the #48 issue you reference, this is the last thing I suggested:

> The MSP430 on the LiFePO4wered/Pi+ doesn't have very strong drivers and I have determined that it normally works with just the I2C pull-ups on the Pi, but if there are any extra pull-ups on the I2C lines, the output low level may not be enough to be read correctly by the Pi.
>
>If you don't have any extra pull-ups on the I2C lines that could cause a logic level problem, it might be interesting as an experiment to add a pull-down of say 22-27K on the SDA line and see if that makes a difference.

After that I didn't get any response anymore.  I don't know if this means it solved the issue or not.  Would you be willing and able to give this a try and report back whether it makes a difference?

DAVe3283, 2022-07-06:
Yeah, but it will take me some time to get a suitable resistor and get it soldered on. The Pi is at my parents' farm, and I only go out there periodically. I will get the ball rolling and report back when I have a conclusion.

DAVe3283, 2022-07-17:
OK, I have attached a 22kΩ pull-down on SDA to ground. Nothing on SCL. Going to hammer it tonight (read the CLI every ~15 seconds) and see what happens.

Starting registers:

```
I2C_REG_VER = 4
I2C_ADDRESS = 67
LED_STATE = 1
TOUCH_STATE = 0
TOUCH_CAP_CYCLES = 20
TOUCH_THRESHOLD = 12
TOUCH_HYSTERESIS = 2
DCO_RSEL = 14
DCO_DCOMOD = 177
VIN = 5093
VBAT = 3563
VOUT = 4952
VBAT_MIN = 2854
VBAT_SHDN = 2952
VBAT_BOOT = 3152
VOUT_MAX = 3889
VIN_THRESHOLD = 4498
VOFFSET_ADC = 34
AUTO_BOOT = 4
WAKE_TIME = 0
SHDN_DELAY = 65
AUTO_SHDN_TIME = 65535
PI_RUNNING = 1
CFG_WRITE = 0
```

DAVe3283, 2022-07-17:
No luck. The LiFePO4wered-Pi shut down after ~16 hours while no one was home.

From my logs:

- After 50 minutes, PI_RUNNING went from 1 to 135 (0x87). Then after another 5 hours it went to 255 (0xFF) for 2 hours, and reverted to 145 (0x87). It bounced a bit more before the Pi shut down.
- After 6 hours, WAKE_TIME went from 0 to 65415 (0xFF87).
- The rest of the parameters I logged stayed the same this time, no weird voltage offset corruption.

So 22kΩ doesn't help, sadly. I will stop using the CLI again, it seems to work OK so long as I don't read from it.