xorbit/LiFePO4wered-Pi #65: LiFePO4wered/Pi+ device will not power up

twadam, 2024-01-13:
After starting with a base Raspberry Pi OS, I followed the GitHub instructions on the code screen to download and install the software.  I have made no changes to any default settings.  After physically installing the LiFePO4wered/Pi+ header with 18650 battery, and only powering the LiFePO4wered device (not the PI) , it seems pushing the physical button for 2 seconds does not do anything.  The battery does have 3.4 volts.  Did I do something wrong or miss a step somewhere ?

twadam, 2024-01-13:
I'm an idiot, nevermind.  I 'think' what I might have done was blindly followed the install instructions including the "make all USE_SYSTEMD=0", which I believe essentially eliminates the daemon. 

At any rate I removed the battery for 10 minutes and removed the entire LiFePO4wered-Pi directory and walked through the GIT instructions again.  This time omitting the 'make all USE_SYSTEMD=0' step.  End result, it powered up perfectly after holding the button for 2 seconds!  Now it's time to play around :-)

xorbit, 2024-01-14:
Hi I'm glad this was one of those issues that solved itself! 🙂 The daemon needs to be running for the device to work correctly, but you'll know for sure that this is the case if the system successfully runs for more than 5 min since the boot timeout would kick in and turn power off if the daemon wasn't running.

closed 2024-01-17 by xorbit