xorbit/LiFePO4wered-Pi #66: AUTO_BOOT_VIN_SMART2 if system shutdown by shell?

Commifreak, 2024-02-23:
Hey.

First, this product is awesome! Keep up the good work!

How is `AUTO_BOOT_VIN_SMART2` behaving when I type `shutdown -h now` inside a shell? I thought, if I power cycle the main input, it will come back up, but it does nothing. is `user-requested shutdown` only via button press?

Thanks in advance!

xorbit, 2024-02-23:
Odd.  I just verified and I know this is not what you want to hear but "it works on my system". :)
AUTO_BOOT_VIN_SMART2 is supposed to work with all shutdown methods.  The only boot criterion is that power needs to be removed and then come back to trigger the boot.

Commifreak, 2024-02-23:
I will bring my pi back to my lab tomorrow and will test this more intense.

Commifreak, 2024-02-24:
That are my next tests. I pick up the unit now.

i set it to 7 and saved it to memory with magic write method before I did one test in prod environmemt. I post my findings within this day.

Commifreak, 2024-02-24:
Okay. I have the unit here now.

I can confirm, that AUTO_BOOT has stopped completely for button-triggered-shutdown AND shutdowns like `shutdown -h now`.

Pi shuts itself down fine (button-method)
Power from LifePo4wered disconnected
waiting 5 secs
power connected
nothing happens (except the CHG LED turns on for some secs).

the cli tool reported params:

```
I2C_REG_VER = 7
I2C_ADDRESS = 67
LED_STATE = 1
TOUCH_STATE = 0
TOUCH_CAP_CYCLES = 0
TOUCH_THRESHOLD = 12
TOUCH_HYSTERESIS = 2
DCO_RSEL = 14
DCO_DCOMOD = 155
VIN = 5153
VBAT = 3424
VOUT = 4993
IOUT = 438
VBAT_MIN = 2850
VBAT_SHDN = 2950
VBAT_BOOT = 3150
VOUT_MAX = 3500
VIN_THRESHOLD = 4498
IOUT_SHDN_THRESHOLD = 0
VBAT_OFFSET = 0
VOUT_OFFSET = 41912
VIN_OFFSET = 419766
IOUT_OFFSET = 0
AUTO_BOOT = 7
WAKE_TIME = 0
SHDN_DELAY = 40
AUTO_SHDN_TIME = 65535
PI_BOOT_TO = 300
PI_SHDN_TO = 120
RTC_TIME = 1708767378
RTC_WAKE_TIME = 0
WATCHDOG_CFG = 0
WATCHDOG_GRACE = 20
WATCHDOG_TIMER = 20
PI_RUNNING = 1
CFG_WRITE = 0
```

What could I test now?

xorbit, 2024-02-24:
Are both the hardware and the software fairly recent?  Can you tell me the hardware revision and when you installed/updated the software last? (To the year is good enough)

Commifreak, 2024-02-24:
The software was built from master 3 days ago and the hardware was bought one week ago from 
https://www.welectron.com/CrowdSupply-LiFePO4wered-Pi-18650-Battery-Non-stackable-Header

If you tell me where to exactly check HW Rev. I can check it. Underway currently so I couldnt look at the board :)

Commifreak, 2024-02-24:
The board has the micro-usb port (i read there is a usb-c one?) and is labeled with „Copyright (c) 2018“. Not sure if this means anything. No „Rev. x“ on the board

xorbit, 2024-02-25:
If it shows no revision number next to the big inductor, it must be a revision 4 or older.  Unfortunately `AUTO_BOOT_VIN_SMART2` is a later addition to the firmware, and again unfortunately this was undocumented for a while and I can't pinpoint for sure when I started shipping units that support this option.

[I realize in the future I should do better with tracking or being able to tell what firmware ships when.  `AUTO_BOOT_VIN_SMART2` was a bit of a weird one because it started as a special request by a customer who then never confirmed that it did what they were looking for, eventually I just decided that it was useful the way it was whether that's what they wanted or not.]

The firmware will allow you to set auto boot values that it doesn't recognize, it will just not do anything and behave like no auto boot (value 0).

I have no relationship with the seller you link to, they're not an "authorized reseller" so to speak.  They must have just bought stock to resell at some point.  Stock currently sold through Mouser is revision 7 for the non-stackable header version which has a micro-B connector and revision 8 for stackable header version which has a Type-C connector.
https://www.mouser.com/c/?q=cs-lifepo

Commifreak, 2024-02-25:
Ahhh there we go! Must be blind:

its a Rev 5 board! Thats not capable I think?

Commifreak, 2024-02-25:
I believe I jzst go with mode 3 here then, thats totally sufficient :)

Thanks for your help and explanation!

closed 2024-02-25 by Commifreak