xorbit/LiFePO4wered-Pi #67: What is the target battery voltage/loading curve?

Commifreak, 2024-03-08:
Hey,

LiFePo4 batteries does not play well with temps below 5° / cold temps in general. I observed a small voltage drop when the night is cold. However, this night, the voltage got below the shutdown threshold so the Pi had some ordered reboot cacles due to LiFePO4wered-Pi detected that the voltage is low.

I checked my normal batter voltage history and the voltage is a bit low. I dont know when my copy of the battery was produced: Is it worth swapping the battery to a new one to get better voltage?

[image]

The Pi is setup outside, so the next winter will bring the voltage down again. Any other advise for my setup? Another kind of battery?

Thanks!

xorbit, 2024-03-12:
I'm not sure, do you think your setup is power positive enough during winter to significantly raise charge level?  That's usually the first issue to check, during winter solar panels tend to produce way below their advertised power levels.
The nominal battery voltage is 3.2V.
I think K2 produces some batteries that are specified to be used in low temperatures: https://k2battery.com/product/k2-energy-3-3v-1-35ah-lithium-ion-phosphate-power-cells-lfp18650p-1350/

Commifreak, 2024-03-12:
> I'm not sure, do you think your setup is power positive enough during winter to significantly raise charge level?

Power comes from a 230V AC to 5V DC power supply

I will see if I can optimize some things.

at which voltage does the Hat stop charging the battery?

xorbit, 2024-03-12:
Oh sorry I misunderstood, I thought it was solar powered.  It stops charging at 3.6V.

Commifreak, 2024-03-12:
Thats interesting because it never got this far later in that graph?

xorbit, 2024-03-12:
Agreed, very odd.  Now if the battery had gone bad, I would expect it to lose capacity and it should reach 3.6V quicker when external power is present.  Which makes me think somehow it's getting insufficient input power.  Which board revision do you have?  Are you monitoring Vin by chance?

Commifreak, 2024-03-12:
Yea, just after my last post I checked my VIn history. And somehow it dropped to 4,8V. Will check that.

xorbit, 2024-03-12:
That's on the low side, the system may be limiting charge current to prevent the input voltage from going lower.
Revision 7 and 8 hardware use a charge chip that deals with this better than older revisions.

Commifreak, 2024-03-18:
I believe the cable is bad. The power supply delivers 5.3V now and VIN only shows 5.03V now. Anyway, case closed.

closed 2024-03-18 by Commifreak