xorbit/LiFePO4wered-Pi #69: Run Script before Shutdown

mmoollllee, 2024-05-28:
Is there a easy way to send a last log message to a server before shutting down?

xorbit, 2024-05-28:
Probably the easiest would be to use something like suggested here:

https://unix.stackexchange.com/questions/39226/how-to-run-a-script-with-systemd-right-before-shutdown

This should work whichever way a shutdown is triggered, LiFePO4wered or something else.

mmoollllee, 2024-05-30:
Thanks a lot!
I came up with the following while handling the SIGTERM signal in the running python script:
```
[Unit]
Description=Log Shutdown to server
After=network.target

[Service]
Type=oneshot
ExecStart=/bin/true
ExecStop=/bin/sh -c 'cat /home/user/app/.pid | xargs kill'
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
```

closed 2024-05-30 by mmoollllee